renode/renode-docs #6: Can't get iceprog to work in the Fomu tutorial

tcal-x, 2020-06-06:
Hi @mateusz-holenko , 

I tried doing the tutorial tutorials/fomu-example.rst, and had some trouble with `iceprog`.  I know that with something like this, where external code may change (iceprog), and there are multiple versions of Fomu, it's difficult to make sure that everything works.   I'll just share my notes:

* The tutorial instructs to load the bitstream before plugging in the fomu
* I needed to install libftdi-dev to get ftdi.h for building iceprog (maybe because I'm using gLinux instead of Ubuntu 18.04)
* `iceprog` not working -- it tries (vendor_id 0x0403, device_id 0x6010 or 0x6014), but maybe it should be  `iceprog -d i:0x1209:0x5bf0 foboot-bitstream.bin`
* But even when specifying correct vendorId:productID, it doesn’t work.
* My Fomu is “Product: Fomu PVT running DFU Bootloader v2.0.3” (according to dmesg)
* Maybe need udev update (see Fomu workshop)?
* Even after changing udev settings according to the Fomu workshop, `iceprog` still doesn’t work

I'll see if anyone on the #tomu channel has any ideas.

tcal-x, 2020-06-07:
The error I get with iceprog is:
```
Init..
Can't find iCE FTDI USB device (device string i:0x1209:0x5bf0).
ABORT.
```
and, I got this info from Xobs:
```
Fomu doesn't have an FTDI chip -- it IS the USB device.
As such, it uses standard USB DFU.
Use `dfu-util`.
```

tcal-x, 2020-06-07:
Wait, is the step using `iceprog` just loading the Fomu "bootloader"?   That is usually already installed; at least, it was on the Fomu that I got from Tim A.

mateusz-holenko, 2020-06-08:
Hi @tcal-x!

> I tried doing the tutorial tutorials/fomu-example.rst, and had some trouble with `iceprog`. I know that with something like this, where external code may change (iceprog), and there are multiple [...]

You are right, this needs to be updated. Plugging in the fomu should be definiately done *before* trying to load the bitstream ;)

> * I needed to install libftdi-dev to get ftdi.h for building iceprog (maybe because I'm using gLinux instead of Ubuntu 18.04)

`iceprog` indeed links to `libftdi`, so you need to have headers available in your system. I'll update the section and include the list of prerequisites together with a note on how to install them (at least on some distros).

> * `iceprog` not working -- it tries (vendor_id 0x0403, device_id 0x6010 or 0x6014), but maybe it should be  `iceprog -d i:0x1209:0x5bf0 foboot-bitstream.bin`
> * But even when specifying correct [...]

I'll need to verify this locally. Have you tried using `sudo` to run `iceprog`?

> Wait, is the step using `iceprog` just loading the Fomu "bootloader"? That is usually already installed; at least, it was on the Fomu that I got from Tim A.

Yes, that's correct - this might not be obvious for the reader (and we need to rephrase this part of tutorial), but by loading a bitstream we understand loading the Fomu "bootloader". You are right that the manufactured fomus should already be pre-installed with it, so this step should be made optional.

mateusz-holenko, 2020-06-08:
> Yes, that's correct - this might not be obvious for the reader (and we need to rephrase this part of tutorial), but by loading a bitstream we understand loading the Fomu "bootloader". You are right [...]

Now that I think about it - in order to upload the Fomu "bootloader" on a fresh device (i.e., without any previous bootloader) you need to have an external programmer for Fomu (e.g., https://github.com/antmicro/fomu-programmer) - you cannot do this directly over USB. Once you have a board with a bootloader you can exchange it using `dfu-util` and don't need `iceprog` anymore.
In this context we probably should drop the whole section related to iceprog from the tutorial and assume the user has a proper FOMU with a bootloader already built-in.

tcal-x, 2020-06-08:
Hi Mateusz,

Yes, your last comment matches my understanding of how Fomu works.

I recall reading in one section of the Fomu workshop that you needed the updated bootloader, but it was just for that one particular sub-demo.   Do you know if this Renode demo needs the updated bootloader (v2.0.3)?   If not, then you can just skip the section about loading/updating the bitstream entirely, and assume that what's installed will work.

I *did* have the older bootloader originally, but have since updated it.   I guess it would be possible to get an old version from foboot and install an older version if we want to test if it works with Renode.

-Tcal

mateusz-holenko, 2020-06-16:
@tcal-x We rephrased the tutorial to make things clearer - could you take a look again and comment on the current version?

tcal-x, 2020-06-17:
Thanks Mateusz!

Yes, definitely an improvement.

* Does `iceprog` actually work for you?   I had the impression from xobs that it could never work for Fomu, and that you had to use `dfu-util`.
* Small typo "Now you can control the HW LED **form** ...."
* Out of curiosity, could the exact same Zephyr image run completely on HW on Fomu, and could the user connect to it over the USB?

mateusz-holenko, 2020-06-17:
> * Does `iceprog` actually work for you?   I had the impression from xobs that it could never work for Fomu, and that you had to use `dfu-util`.

Yes, it works, *but* you need to use an external programming board (i.e., you don't use the FOMU USB interface to program it) - take a look at: https://github.com/antmicro/fomu-programmer.

> * Small typo "Now you can control the HW LED **form** ...."

Thanks for spotting, it's fixed now.

> * Out of curiosity, could the exact same Zephyr image run completely on HW on Fomu, and could the user connect to it over the USB?

FOMU has a limited amount of memory, so probably only a minimal version of Zephyr (e.g., no colors in shell) will fit. We have some experience with running Zephyr on ice40, but not on FOMU itself.
Another thing is that FOMU does not provide any UART interface - you'll need to tunnel it through USB and I'm not sure if that is currently supported in Zephyr.

tcal-x, 2020-07-08:
Closing, sorry I overlooked this.

closed 2020-07-08 by tcal-x