renode/renode-docs #7: Fomu tutorial: Renode build instructions a bit confusing

tcal-x, 2020-06-07:
Hi @mateusz-holenko , 

In doing the tutorial, I am pointed to https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#building-from-source to build Renode.   However, after installing the prerequisites, cloning the repo, and executing ./build.sh, I don't have 'renode' in my path, and executing ./renode gives an error.

* Is the reader supposed to know that they need to create packages and then install the packages locally?
* Why does the reader need to build from source and not just install the latest package from https://github.com/renode/renode/releases/latest?

I ended up downloading and installing the .deb file for the latest release, and then the demo worked as described.

mateusz-holenko, 2020-06-08:
Hi @tcal-x!

> In doing the tutorial, I am pointed to https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#building-from-source to build Renode. However, after installing the prerequisites, [...]

`renode` binary won't be added to PATH as a result of running `./build.sh`, this script is just for building a local copy of Renode.

Running `./renode`, however, *should* work - what error exactly do you get?
 
> * Is the reader supposed to know that they need to create packages and then install the packages locally?

Installing a locally-built package in the system is an option, but it is not required and we don't expect users to do that. You should be able to run `./renode` script in the cloned repository after doing a build.

> * Why does the reader need to build from source and not just install the latest package from https://github.com/renode/renode/releases/latest?
> 
> I ended up downloading and installing the .deb file for the latest release, and then the demo worked as described.'

Good to hear. Most probably at the time of writing the tutorial the current release did not include all the necessary elements to run this demo and therefore a manual built was necessary.

I'll double-check that the latest release contains everything that is necessary and if so I'll link to the pre-built releases. But again, the locally-built version should work as well (as we provide packages only for selected OSes), so this needs to be verified as well.

tcal-x, 2020-06-08:
Running ./renode, this is what I see:
```
tcal@tcal-glaptop:~/renode/renode$ ./renode
Cannot open assembly '/home/tcal/renode/renode/output/bin/Release/Renode.exe': No such file or directory.
```

mateusz-holenko, 2020-06-08:
It looks like Renode is not built properely - maybe `./build.sh` failed at some point. Could you share an output generated by `./build.sh`?

tcal-x, 2020-06-08:
> It looks like Renode is not built properely - maybe `./build.sh` failed at some point. Could you share an output generated by `./build.sh`?

Sure, I'll do a make clean and rebuild as soon as I get the opportunity.

PiotrZierhoffer, 2020-06-11:
For the record, any new Xamarin version would be good enough. 
I understand you can't use them easily, though.

tcal-x, 2020-06-11:
Yes, in fact Google does have the Xamarin packages in an internal repo.   It was more my laptop and my lack of experience with apt that I couldn't get them installed.   I have found a workaround now though, so I am unstuck.

My issue was that I had both an earlier version of Mono (5.20) from the Xamarin packages, and a later version (6.8) from Debian packages.   Apt really wanted to install the latest version.   Even if I specified version 5.20 for mono-complete, apt wanted to install the 6.8 version of all the dependencies (e.g. mono-runtime), somehow getting confused and then saying "you have held broken packages".   But, my situation may be specific to the Google environment.

So, the question is, will the public at large be able to install the Xamarin version of Mono without difficulty?

tcal-x, 2020-06-12:
I just looked at the Mono packages from my non-work Ubuntu 19.04 laptop, and the situation is reversed.   Xamarin has version 6.8, and Debian has 5.18, so apt will pick Xamarin.   I cloned and compiled Renode with no problems (other than replacing "realpath" with the suggested replacement).

mateusz-holenko, 2020-06-12:
> Making it work on Mono 6.8 "debian edition" might be a better idea @mateusz-holenko @PiotrZierhoffer ? Can we do a dirty test if this is feasible?

@pgielda @tcal-x Working on this.

tcal-x, 2020-07-02:
Hi @mateusz-holenko -- I am trying to rerun on my gLinux laptop.  I purged the Xamarin Mono and just have the default Debian Mono now.

But I'm running into another issue, that I've seen intermittently before -- when I type "./build.sh", it just clones the submodules then exits.   If I rerun, it just says 
```
Required Renode libraries already downloaded. To repeat the process remove .renode_libs_fetched file.
```
and doesn't attempt any building.    "./build.sh -c" gives the same message and doesn't do anything either.

Is there some state that I need to clear to get it to build?

PiotrZierhoffer, 2020-07-02:
@tcal-x this is probably caused by mono compilation failing very early. We should change the behavior here, I think, so it's more verbose.

For debug purposes, can you look at build.sh and remove the redirection to /dev/null at line 156? Link for reference: https://github.com/renode/renode/blob/master/build.sh#L156

Then please paste the output of `mono --version`, `xbuild /version` and of course `./build.sh`

tcal-x, 2020-07-02:
```
$ mono --version
Mono JIT compiler version 6.8.0.105 (Debian 6.8.0.105+dfsg-2 Tue Apr 28 14:28:03 UTC 2020)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       altstack
        Notifications: epoll
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug 
        Interpreter:   yes
        LLVM:          supported, not enabled.
        Suspend:       hybrid
        GC:            sgen (concurrent by default)

$ xbuild /version
XBuild Engine Version 14.0
Mono, Version 6.8.0.105
Copyright (C) 2005-2013 Various Mono authors

$ ./build.sh 
Required Renode libraries already downloaded. To repeat the process remove .renode_libs_fetched file.

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.8.0.105
Copyright (C) 2005-2013 Various Mono authors

Build started 7/2/2020 12:28:55 PM.
__________________________________________________
Project "/home/tcal/r2/renode/lib/cctask/CCTask.sln" (default target(s)):
        Target ValidateSolutionConfiguration:
/home/tcal/r2/renode/lib/cctask/CCTask.sln: error : Invalid solution configuration and platform: "Release|arty".
        Task "Error" execution -- FAILED
        Done building target "ValidateSolutionConfiguration" in project "/home/tcal/r2/renode/lib/cctask/CCTask.sln".-- FAILED
Done building project "/home/tcal/r2/renode/lib/cctask/CCTask.sln".-- FAILED

Build FAILED.
Errors:

/home/tcal/r2/renode/lib/cctask/CCTask.sln (default targets) ->
(ValidateSolutionConfiguration target) ->

        /home/tcal/r2/renode/lib/cctask/CCTask.sln: error : Invalid solution configuration and platform: "Release|arty".

         0 Warning(s)
         1 Error(s)

Time Elapsed 00:00:00.1320950
```

tcal-x, 2020-07-02:
Yeah, that's probably it
```
$ env | grep -i arty
PLATFORM=arty
FULL_PLATFORM=arty
```

tcal-x, 2020-07-02:
Unsetting them seems to have fixed it.
Thanks Piotr!

PiotrZierhoffer, 2020-07-02:
Ha! Never expected that to happen :D I wasn't even aware that envs are analyzed by xbuild.

Glad to hear it helped. I closed the issue by mistake, we'll do it when you confirm everything works as expected on your side.

tcal-x, 2020-07-02:
Yes, it works with default Mono now!

I'll open a new issue in the appropriate repo if you haven't already, for the silent failure due to user env var being set.

tcal-x, 2020-07-02:
I opened https://github.com/renode/renode/issues/106, `Build fails silently due to user env var being set #106`.

Closing this issue.

closed 2020-07-02 by tcal-x