renode/renode-docs #11: Add a chapter on UART host-guest connections

PiotrZierhoffer, 2020-10-06:
As requested by @karlp and other users, host-guest connection is a very common thing to do and we do not have docs for that.

We should write about:

- uart via pseudoterminal
- uart via socket

We need to include details about emitConfig, as it may be confusing

mateusz-holenko, 2020-10-12:
https://github.com/renode/renode-docs/commit/023c976cf4ce7e9c6d38ddaa8108353cc1534e20 adds sections describing the pseudoterminal and socket  host-guest UART connections.

karlp, 2020-10-12:
Um, either you still haven't pushed the right commit, or you closed the wrong issue? that linked commit doesn't relate to this issue, and there's no other recent commits that appear relevant either, suggest this be re-opened?

mateusz-holenko, 2020-10-12:
@karlp you are right - I closed the issue prematurely, sorry for that :(

The commit I linked extends the testing section of the documentation and describes some features we recently added to the `test-renode` tool.

I'm now working on the sections about the pseudoterminal and socket host-guest UART connections and hope to publish them soon.

mateusz-holenko, 2020-10-14:
https://github.com/renode/renode-docs/commit/ddbac814255db80cb3b92bbc16f80dc8de2d36ce adds the UART host-guest integration section covering both the pseudoterminal and the socket-based approach.

closed 2020-10-14 by mateusz-holenko