renode/renode-docs #15: Document how to debug C# modules

xobs, 2020-10-16:
Writing C# modules unlocks a lot of functionality in Renode, however it's very difficult to debug these modules.

For example, in this module:

```cs
using System;
using Antmicro.Renode.Core;
using Antmicro.Renode.Peripherals.Bus;
using Antmicro.Renode.Time;
using Antmicro.Renode.Logging;
using System.Threading;

namespace Antmicro.Renode.Peripherals.Timers
{
    public class SimpleTicker : IDoubleWordPeripheral
    {
        public SimpleTicker(ulong periodInMs, Machine machine)
        {
            machine.ClockSource.AddClockEntry(new ClockEntry(periodInMs, ClockEntry.FrequencyToRatio(this, 1000), OnTick, this, String.Empty));
        }
       // ...
    }
}
```

It's difficult to know:

1. What properties there are on `machine`
2. Where `Machine` is defined (which package does it come from?)
3. What do each of the parameters in that function mean?
4. What is a `ClockSource` and how does it work?
5. What other properties are there in `Antmicro.Renode.Peripherals`?
6. What is the interface required to implement `IDoubleWordPeripheral`?
7. Does this even compile?

Many of these questions could be answered by connecting this `.cs` file to an IDE of some sort. I'm not sure how `2` could be answered, because I gather that's a feature of C#. `7` can only be answered by restarting Renode.

With C# modules being documented, it would also be nice to know how to implement modules and debug issues within them.

xobs, 2020-10-27:
I'm starting to understand how to integrate this with Omnisharp.

I now have this:

[image]

Note the gray up top, indicating I'm not using particular modules. Also, note that mouseover gives information on arguments.

I get full autocomplete:
[image]

And if I control-click on a symbol, it brings me to a useful screen describing it:

[image]

In order to get this, create a `.csproj` file somewhere in the directory and point it to Renode. In my case, this file looks like:

<details>
<summary>Click to show `peripherals.csproj`</summary>

```xml
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <TargetFramework>netstandard2.0</TargetFramework>
    <PlatformTarget>AMD64</PlatformTarget>
  </PropertyGroup>

  <ItemGroup>
    <Reference Include="AdvancedLoggerViewerPlugin">
      <HintPath>..\..\..\..\Software\Renode\bin\AdvancedLoggerViewerPlugin.dll</HintPath>
    </Reference>
    <Reference Include="AntShell">
      <HintPath>..\..\..\..\Software\Renode\bin\AntShell.dll</HintPath>
    </Reference>
    <Reference Include="AsyncIO">
      <HintPath>..\..\..\..\Software\Renode\bin\AsyncIO.dll</HintPath>
    </Reference>
    <Reference Include="CookComputing.XmlRpcV2">
      <HintPath>..\..\..\..\Software\Renode\bin\CookComputing.XmlRpcV2.dll</HintPath>
    </Reference>
    <Reference Include="cores-arm">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-arm.dll</HintPath>
    </Reference>
    <Reference Include="cores-arm-m">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-arm-m.dll</HintPath>
    </Reference>
    <Reference Include="cores-i386">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-i386.dll</HintPath>
    </Reference>
    <Reference Include="cores-ppc">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-ppc.dll</HintPath>
    </Reference>
    <Reference Include="cores-ppc64">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-ppc64.dll</HintPath>
    </Reference>
    <Reference Include="cores-riscv">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-riscv.dll</HintPath>
    </Reference>
    <Reference Include="cores-riscv64">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-riscv64.dll</HintPath>
    </Reference>
    <Reference Include="cores-sparc">
      <HintPath>..\..\..\..\Software\Renode\bin\cores-sparc.dll</HintPath>
    </Reference>
    <Reference Include="CxxDemangler">
      <HintPath>..\..\..\..\Software\Renode\bin\CxxDemangler.dll</HintPath>
    </Reference>
    <Reference Include="Dynamitey">
      <HintPath>..\..\..\..\Software\Renode\bin\Dynamitey.dll</HintPath>
[... 75 lines cut ...]
    </Reference>
    <Reference Include="UI">
      <HintPath>..\..\..\..\Software\Renode\bin\UI.dll</HintPath>
    </Reference>
    <Reference Include="VerilatorPlugin">
      <HintPath>..\..\..\..\Software\Renode\bin\VerilatorPlugin.dll</HintPath>
    </Reference>
    <Reference Include="WiresharkPlugin">
      <HintPath>..\..\..\..\Software\Renode\bin\WiresharkPlugin.dll</HintPath>
    </Reference>
    <Reference Include="Xwt">
      <HintPath>..\..\..\..\Software\Renode\bin\Xwt.dll</HintPath>
    </Reference>
    <Reference Include="Xwt.WPF">
      <HintPath>..\..\..\..\Software\Renode\bin\Xwt.WPF.dll</HintPath>
    </Reference>
  </ItemGroup>

</Project>
```
</details>

This obviously has a lot of platform-specific cruft. And I'm not sure on the `<TargetFramework>netstandard2.0</TargetFramework>` value -- isn't it a different version on Windows?

At any rate, this gives me syntax checking, autocomplete, and rudimentary documentation. It also doesn't require me to compile Renode, since I'm just using the binaries directly.

xobs, 2020-10-27:
Also, one can mouseover the arguments to figure out which package they come from. For example, now I know that `Machine` comes from `Antmicro.Renode.Core`:

[image]

xobs, 2020-10-27:
I updated the `peripherals.csproj` file in original comment to make it more cross-platform and less unwieldy. I'd appreciate someone else giving it a try to see if it works for them.

PiotrZierhoffer, 2020-10-27:
Hey @xobs, this is really cool.

We'll take a look internally and then maybe we'd be able to somehow include it in our packages, so that everyone can use it with simple instructions (like `cp this file to your dir, start code, install omnisharp, voila`)