renode/renode-docs #17: Co-simulation with verilator documentation?

mithro, 2020-10-24:
Searching for ["Verilator" in the documentation](https://renode.readthedocs.io/en/latest/search.html?q=Verilator&check_keywords=yes&area=default) returns no links?

Searching for ["Verilog" in the documentation](https://renode.readthedocs.io/en/latest/search.html?q=Verilog&check_keywords=yes&area=default#) also returns no links?

It seems like currently the best documentation for using Renode + Verilator together is actually the Fomu workshop page at https://workshop.fomu.im/en/latest/renode-verilator.html !?

It would probably be good a good start to directly import the page from the Fomu workshop into the Renode documentation.

mithro, 2020-10-24:
FYI - We will be needing this for the simulation of TFLiteMicro "device based" (rather than CFU based) accelerators, so it would help to have decent docs for it...

mithro, 2020-10-24:
FYI - You also have a blog post at https://antmicro.com/blog/2019/09/renode-verilator-hdl-co-simulation/ and a repository at https://github.com/antmicro/renode-verilator-integration

mgielda, 2022-10-17:
https://renode.readthedocs.io/en/latest/tutorials/verilator-cosimulation.html - this chapter has been around for a while. We will be expanding those docs related to some new capabilities we're developing, but I think we can close this particular issue.

closed 2022-10-17 by mgielda