renode/renode-docs #29: Emulating UART port or Device

Nileshhampiholi, 2021-12-01:
Hello , 
I  am trying to emulate a device (or just emulating a port works for me) to write some data to serial port of host device.

I have a software that reads serial  data specifically from - /dev/serial/by-id/ file_name. 

I have tried following approaches. 
1. Tried emulating a stm3f4_discovery
2. emulating arduino nano 33 ble

- Tried emulating a stm3f4_discovery

1. I have stm32cube ide  with code ready to be flashed into the board so that the board or device write 0 and 1 to the serial port
2.  (monitor) emulation CreateUartPtyTerminal "term" "/tmp/uart"
3. mach create
4. machine LoadPlatformDescription @platforms/boards/stm32f4_discovery.rep
5. connector Connect sysbus.uart1 term

And it works until here. 

Further I don't understand what the documentation says -- what does this mean?
(Finally, open the terminal application on your host machine (screen/picocom/PuTTY/etc.) and attach it to the /tmp/uart file. You can interact with it just like with the hardware.)

Attach how? 
-  emulating Arduino nano 33 ble

1. I have Arduino code that write 0 and 1 to the serial port using serial1 as mentioned in the documentation.
2. In renode I run the script s @scripts/complex/arduino_nano/nrf52840_arduinoIDE_integration.resc
3. I flash the code in Arduino ide and see out put in uart analyser

But if I use python and try to read data from serial device connected to the port I don't see anything coming out. .

pgielda, 2021-12-01:
It seems those are actually two questions/issues.

I will try to answer the first one - what the documentation is trying to say -- after executing (2) a special /tmp/uart pty file will be created and you can connect to it using tools like screen or picocom (or any other tool that can handle pseudoterminals).

(I am assuming here you're using Linux.)

With picocom its just:
`picocom /tmp/uart`

but it obviously depends on the tool. I am not sure what tool you are trying to use and if it can handle a custom path like that.

I will let someone else answer the questions about Arduino IDE. @PiotrZierhoffer @mateusz-holenko

Nileshhampiholi, 2021-12-06:
Hello, 
Thank you it works as you said. But still doesn't solve my task. 

How can I achieve this part  - 
I am trying to emulate a device (or just emulating a port works for me) to write some data to serial port of host device.
I have a software that reads serial data specifically from - /dev/serial/by-id/ file_name.

Things I tried is it in right direction? 
Or its there any other way to get this task done.

pgielda, 2021-12-06:
Well I would say just modify the software to accept any path, but I guess you mean you have some closed software that you cannot modify?

In general as far as I remember you can use udevadm to bind a specific port to an id... Not sure if that will work with a pty... if it has to be a "real" uart, then you might be forced to do some linux kernel module tricks to get that to work.

Maybe someone has a better idea here of course...

Nileshhampiholi, 2021-12-06:
Yes I cannot modify the software. I am new to this what kind of kernel manipulations can I do. Is there some documentation I can follow.

PiotrZierhoffer, 2021-12-21:
If you need to open a specific file under `/dev/serial/by-id/`, then one thing you might try is to actually provide this path in Renode:

`emulation CreateUartPtyTerminal "term" "/dev/serial/by-id/some-file-name"`

This would require you to run Renode as root, unfortunately.

It might work, but the terminal we create is fairly basic. If your application does some heavy configuration on the serial device, it might fail with this connection.

Right now this is the only solution we have, unless you have an option to connect via a socket (e.g. via telnet).

But I believe it's worth a try.

PiotrZierhoffer, 2022-10-18:
I'll close this issue. Please feel free to open a new one if needed.

closed 2022-10-18 by PiotrZierhoffer