renode/renode-docs #35: Missing comma in logging basics

IgorErin, 2024-04-06:
Hello. I've tried using [this](https://github.com/renode/renode-docs/blob/342eb5b74bbb874354201cefbc9838658677ed50/source/basic/logger.md?plain=1#L176) and [this](https://github.com/renode/renode-docs/blob/342eb5b74bbb874354201cefbc9838658677ed50/source/basic/logger.md?plain=1#L168) snippet of code. It only worked after I added a comma between the range boundaries. Should the comma be there?

pgielda, 2024-04-06:
Depends what you want to do...
sysbus SilenceRange <A B> == sysbus SilenceRange <A,A+B-1>

pgielda, 2024-04-06:
So A is address and B is size of the region if you're using space in between. If you use comma, the second parameter is the offset of the end of the range, not size.

closed 2024-04-06 by pgielda